AlessandroAU/Chorus32-ESP32LapTimer #1: Wifi drop

mmoustache, 2018-12-01:
Hi, after flashing scetch in Network manager i see Chorus LapTimer wifi. Connecting, and after 2 second my connection is dropping and phone trying another wifi net. Im Using DOIT ESP32 DEVKIT 1.
My flash log is below:

 Build options changed, rebuilding all
Archiving built core (caching) in: /var/folders/vh/7yvqz7hs4mb3hx6h7yhhwkgh0000gn/T/arduino_cache_787905/core/core_esp32_esp32_esp32_PSRAM_disabled,PartitionScheme_min_spiffs,FlashMode_qio,FlashFreq_80,FlashSize_4M,UploadSpeed_115200,DebugLevel_none_c000cfe2de910e434d508ffe474b5bc9.a
Sketch uses 1328918 bytes (67%) of program storage space. Maximum is 1966080 bytes.
Global variables use 82732 bytes (25%) of dynamic memory, leaving 244948 bytes for local variables. Maximum is 327680 bytes.
esptool.py v2.3.1
Connecting........__
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 8221.8 kbit/s)...
Hash of data verified.
Compressed 15072 bytes to 9733...

Writing at 0x00001000... (100 %)
Wrote 15072 bytes (9733 compressed) at 0x00001000 in 0.9 seconds (effective 139.9 kbit/s)...
Hash of data verified.
Compressed 1329056 bytes to 779928...

Writing at 0x00010000... (2 %)
Writing at 0x00014000... (4 %)
Writing at 0x00018000... (6 %)
Writing at 0x0001c000... (8 %)
Writing at 0x00020000... (10 %)
Writing at 0x00024000... (12 %)
Writing at 0x00028000... (14 %)
Writing at 0x0002c000... (16 %)
Writing at 0x00030000... (18 %)
Writing at 0x00034000... (20 %)
Writing at 0x00038000... (22 %)
Writing at 0x0003c000... (25 %)
Writing at 0x00040000... (27 %)
Writing at 0x00044000... (29 %)
Writing at 0x00048000... (31 %)
Writing at 0x0004c000... (33 %)
Writing at 0x00050000... (35 %)
Writing at 0x00054000... (37 %)
Writing at 0x00058000... (39 %)
Writing at 0x0005c000... (41 %)
Writing at 0x00060000... (43 %)
Writing at 0x00064000... (45 %)
Writing at 0x00068000... (47 %)
Writing at 0x0006c000... (50 %)
Writing at 0x00070000... (52 %)
Writing at 0x00074000... (54 %)
Writing at 0x00078000... (56 %)
Writing at 0x0007c000... (58 %)
Writing at 0x00080000... (60 %)
Writing at 0x00084000... (62 %)
Writing at 0x00088000... (64 %)
Writing at 0x0008c000... (66 %)
Writing at 0x00090000... (68 %)
Writing at 0x00094000... (70 %)
Writing at 0x00098000... (72 %)
Writing at 0x0009c000... (75 %)
Writing at 0x000a0000... (77 %)
Writing at 0x000a4000... (79 %)
Writing at 0x000a8000... (81 %)
Writing at 0x000ac000... (83 %)
Writing at 0x000b0000... (85 %)
Writing at 0x000b4000... (87 %)
Writing at 0x000b8000... (89 %)
Writing at 0x000bc000... (91 %)
Writing at 0x000c0000... (93 %)
Writing at 0x000c4000... (95 %)
Writing at 0x000c8000... (97 %)
Writing at 0x000cc000... (100 %)
Wrote 1329056 bytes (779928 compressed) at 0x00010000 in 69.0 seconds (effective 154.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.0 seconds (effective 1452.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

AlessandroAU, 2018-12-01:
Hi, you upload the serial log when you try to connect to WiFi?

mmoustache, 2018-12-02:
Hi, serial log says that, maybe it happend beacuase i justa flashed ESP32 and not connected receivers?

Ready! Open http://esp32-webupdate.local in your browser
Ready
IP address: 0.0.0.0
setModuleChannelBand
0,4
32
a = 5740
setModuleChannelBand
2,4
34
a = 5780
setModuleChannelBand
4,4
36
a = 5820
setModuleChannelBand
6,4
38
a = 5860
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x400dd628  PS      : 0x00060034  A0      : 0x800817e4  A1      : 0x3ffc0bd0  
A2      : 0x00000000  A3      : 0x20000000  A4      : 0x00000400  A5      : 0x3ffc5bb4  
A6      : 0x3ffc4d40  A7      : 0xffffffff  A8      : 0x80081290  A9      : 0x3ffc0bb0  
A10     : 0x00000000  A11     : 0xffffffff  A12     : 0x3ffd69ec  A13     : 0x000000ff  
A14     : 0x3ffc63c0  A15     : 0x00060a23  SAR     : 0x00000014  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x40082f73  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400dd628

Backtrace: 0x400dd628:0x3ffc0bd0 0x400817e1:0x3ffc0bf0 0x40081cd9:0x3ffc0c10 0x40082f70:0x00000000

Rebooting...

AlessandroAU, 2018-12-02:
Seems to be crashing during the ISR loop when reading ADCs, should not be happening. I will have a look.

AlessandroAU, 2018-12-07:
Appears to be a conflict between wifi and bluetooth stacks, fixed for now by disabling BLE by default in latest commit. BLE can be re-enabled by uncommenting compiler option //#define BluetoothEnabled

closed 2018-12-07 by AlessandroAU