AlessandroAU/Chorus32-ESP32LapTimer #2: File ESPmDNS.h is missing for compile/upload

mottex, 2019-02-03:
I try to upload the laptimer code, but arduino ide - gives the following error message:

ESP32LapTimer:7:21: error: ESPmDNS.h: No such file or directory

compilation terminated.

exit status 1
ESPmDNS.h: No such file or directory

Thanks for your help

AlessandroAU, 2019-02-03:
You will need to install all the libraries required. You should be able to do it through the library manager 
[image]

mottex, 2019-02-03:
[image]

Unfortunately it is not shown.... any other suggestions?

AlessandroAU, 2019-02-03:
I would do I fresh install of the esp32 files through board manager to ensure you're not having issues because of old code

mottex, 2019-02-03:
Thanks for your help! I have internet connection problems, which caused the above message. 

Now, i got the message - comms.h -
[image]
 is missing. Do i have to include an other library? (the File comms.h is in the project folder).

AlessandroAU, 2019-02-04:
You need to make sure that all the files are in the same folder and that the folder is called "ESP32LapTimer" ie same as the main arduino .ino file.

mottex, 2019-02-04:
Thanx allessandro! Can be closef

closed 2019-02-04 by mottex