AlessandroAU/Chorus32-ESP32LapTimer #3: Hardware

LuisDiazUgena, 2019-02-05:
Hi! 

I was wondering if you can share a link to the board that you're using to built this.

TY.

AlessandroAU, 2019-02-07:
Hi, most ESP32 boards will work. I used an OLED board but I would recommend a standard nodeMCU ESP-32s as it will probably become the standard.

LuisDiazUgena, 2019-02-07:
Ok, no problem. Will take a look at which boards are available and choose one. I think maybe it will be interesting to make a custom pcb board to solder on the ESP32 IC and the VRX. Also, by adding some simple power management stuff, could be a nice project to work with.

Also, could be nice to keep info about how to create and wire up the project.

AlessandroAU, 2019-02-08:
Hi, I added a wiring diagram for the NodeMCU ESP32s
https://github.com/AlessandroAU/Chorus32-ESP32LapTimer/blob/master/wiring.png

closed 2019-02-08 by LuisDiazUgena