AlessandroAU/Chorus32-ESP32LapTimer #4: Connection Diagram needed

shirishyc, 2019-02-07:
Hi

I want to use an OLED board for this Project.
So can you provide Connection diagram for connecting ESP32 and RX5808 module

Waiting for your reply

Thanks in Advance
SHIRISH

LuisDiazUgena, 2019-02-07:
Regarding the oled:

For connecting the OLED you shoul use the rigth port on the node MCU, I think most of oled uses SCL/SDA to interface with the OLED. 

Google for "your_board_name_here pinout" and look for SCL/SDA.

AlessandroAU, 2019-02-08:
Hi, recommend you don't use the esp32 oled board because not all the ADC pins are broken out. At the moment the oled does nothing. The NodeMCU ESP32s is easier to wire up. I have added a wiring diagram to the main page. 

https://github.com/AlessandroAU/Chorus32-ESP32LapTimer/blob/master/wiring.png

shirishyc, 2019-02-08:
Thanks a lot for updating Git repo with wiring diagram.
Will work accordingly..

shirishyc, 2019-02-10:
Hi,
I have compiled uploaded into ESP32.
But having small question, What file need to upload in WebServer.
Im unble to figure out.. Can you let me know sample file for the same.. or can you upload into repo.
Im attaching screenshot for ref.

Waiting for reply
Thanks
SHIRISH

[image]

AlessandroAU, 2019-02-10:
You don't need to upload anything, that is for a future version where you will be able to update the firmware by uploading a file.

It should be working, check to see if you can make a connection with the app.

shirishyc, 2019-02-10:
Ok Great Thanks.
Im able to connect to App. But unable to add pilots name..
is I'm missing something to do..

AlessandroAU, 2019-02-10:
Can you show me the serial console in Arduino when you connect to the board from the app?

AlessandroAU, 2019-02-12:
Did you get it working?

shirishyc, 2019-02-13:
Yes.. Its Perfectly Working Fine..
I just mess with some Wires thts why Unable to do.!
But Now working Fine.

Thanks for Awesome Code.

1que Can We run web server from ESP32

closed 2019-02-13 by shirishyc

AlessandroAU, 2019-02-13:
This is possible, but it would be alot of work and web design is not my area of expertise. I think it is too hard for the moment.

marcfon, 2020-04-30:
In case someone is searching for the wiring diagram... The link has changed to this:  https://github.com/AlessandroAU/Chorus32-ESP32LapTimer/blob/master/img/wiring.png