AlessandroAU/Chorus32-ESP32LapTimer #23: Unable to connect to wifi configurator

LuisDiazUgena, 2019-03-07:
Hi!

I was trying to use this project and I'm not being able to connect to the wifi configurator.

I've flashed the code with no issues. Also flashed href=\"https://github.com/me-no-dev/arduino-esp32fs-plugin pluging and uploaded esp32 sketch data.

I can connect to wifi AP, but android doesn't ask me to sign in the network. I've looked into the code, and the wifi AP seems to be 192.168.1.4, so I've tried to log using chrome, not luck again.

Here is my serial monitor output.
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10088
load:0x40080400,len:6380
entry 0x400806a4
[V][esp32-hal-i2c.c:1483] i2cInit(): num=0 sda=21 scl=22 freq=0
[V][esp32-hal-i2c.c:1677] i2cSetFrequency(): freq=100000Hz
[V][esp32-hal-i2c.c:1692] i2cSetFrequency(): cpu Freq=240Mhz, i2c Freq=100000Hz
[V][esp32-hal-i2c.c:1698] i2cSetFrequency(): Fifo delta=1
[V][esp32-hal-i2c.c:1677] i2cSetFrequency(): freq=700000Hz
[V][esp32-hal-i2c.c:1692] i2cSetFrequency(): cpu Freq=240Mhz, i2c Freq=700000Hz
[V][esp32-hal-i2c.c:1698] i2cSetFrequency(): Fifo delta=5
Booting....
EEPROM LOADED
6
6
[V][esp32-hal-i2c.c:1483] i2cInit(): num=0 sda=21 scl=22 freq=0
[V][esp32-hal-i2c.c:1677] i2cSetFrequency(): freq=701754Hz
[V][esp32-hal-i2c.c:1692] i2cSetFrequency(): cpu Freq=240Mhz, i2c Freq=701754Hz
[V][esp32-hal-i2c.c:1698] i2cSetFrequency(): Fifo delta=5
off
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 14 - AP_STOP
[E][WiFiSTA.cpp:211] begin(): connect failed!
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 13 - AP_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 13 - AP_START
on
HTTP server started
[E][WiFiClient.cpp:264] setOption(): 9
setModuleChannelBand
0,4
32
a = 5740
setModuleChannelBand
2,4
34
a = 5780
setModuleChannelBand
4,4
36
a = 5820
setModuleChannelBand
6,4
38
a = 5860
setModuleChannelBand
0,0
0
a = 5658
setModuleChannelBand
7,0
7
a = 5917
Starting ADC reading task on core 1
Task created...
E (10695) event: mismatch or invalid event, id=63
E (10696) event: default event handler failed!
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 15 - AP_STACONNECTED
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 15 - AP_STACONNECTED
dhcps: send_offer>>udp_sendto result 0
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 17 - AP_STAIPASSIGNED

```

JyeSmith, 2019-03-07:
The address is http://192.168.4.1/

See how you go with that :)

LuisDiazUgena, 2019-03-07:
Nope, same behaviour :(

AlessandroAU, 2019-03-07:
Hi Luis,

Sorry this was my fault, was trying to debug a separate issue and left some code in that shouldn't have been there. I just fixed it, try it again :)

LuisDiazUgena, 2019-03-08:
Yes! It's working now :) Now it's asking to sign in the network and I can see the webpage.

Still not able to see the site using the url 192.168.4.1

If you leave the page, it's imposible to see it until you turn off and on wifi connection.

closed 2019-03-08 by LuisDiazUgena