AlessandroAU/Chorus32-ESP32LapTimer #28: Fast scanning to reduce vrx

LuisDiazUgena, 2019-03-08:
Would be possible to implement a mode in which esp32 makes the vrx jumps through channels really quick?

That way it would be possible to reduce the number of vrx to complete the project.

JyeSmith, 2019-03-08:
Yes, but no :)

The RX5808 module takes about 30ms to settle and output a sensible RSSI after changing channels.  For a quad moving at 100km/h in 30ms it moves 83cm, so that now becomes your timing resolution.  Do this for 6 channels and the quads RSSI is now only being sampled every 5m.  At that distance a quad could pass through the timing gate and be missed completely.

LuisDiazUgena, 2019-03-11:
That's a shame. Will be a great option to reduce the cost and size.. but if it's not possible, it's not possible. Thanks

closed 2019-03-11 by LuisDiazUgena

bseishen, 2019-03-30:
Both TBS and Immersions personal tracker offerings both use one module. I think this is worth perusing even loss of resolution. Might not be a good solution for large races, but practice days 1-4 pilots or even tinywhoop might be a good idea. 
According to Immersions laprf app:
1 Pilot - 30ms/160kmh
2 - 60/90
3- 90/75
4-120/50