AlessandroAU/Chorus32-ESP32LapTimer #31: Changing channel from webserver

LuisDiazUgena, 2019-03-11:
Hi!

I started to create the interface for updating band and channel from webserver:

[image]

The select items are showing bands and channels and will be showed with the parsed value from the configuration `data.NumRXs`. So, ie: if data.NumRXs = 4, only 4 VRX configuration will be showed. 

Channels can be chossed from 1 to 8 and bands ussing their "names":

[image]

As I click the update [button](https://github.com/AlessandroAU/Chorus32-ESP32LapTimer/blob/76596c2e07e1e9246a0b7944631699c094e3b582/ESP32LapTimer/data/index.html#L106) on index, the url will be:

`updateGeneral?NumRXs=7&band1=R&channel1=1&band2=R&channel2=2&band3=R&channel3=3&band4=R&channel4=4&band5=R&channel5=5&band6=R&channel6=6`

Where each band and channel are sended ussing the same pattern. 

I assume  data transfer is going to be recieved at [this line in webserver.ino](https://github.com/AlessandroAU/Chorus32-ESP32LapTimer/blob/76596c2e07e1e9246a0b7944631699c094e3b582/ESP32LapTimer/WebServer.ino#L259)

And needed to be stored in the eepromsettings. I think they need to be stored in RXBand, RXChannel and RXFrequencies.

I don't have the project with me, so I can't test it out. but will do during this week and let you know. 

If you got any suggestions, please, feel free to contact me.

JyeSmith, 2019-03-11:
Nice.

My preference would be for buttons like the app but can live with drop down boxes.

LuisDiazUgena, 2019-03-11:
Can you post an image to show how it's this done in the app?

JyeSmith, 2019-03-11:
You can see a screen shot here.

https://play.google.com/store/apps/details?id=app.andrey_voroshkov.chorus_laptimer

LuisDiazUgena, 2019-03-12:
Thanks @JyeSmith, I think that's a good idea as well. Will try to implement it, but for now I think it's ok with the dropdown.

AlessandroAU, 2019-07-07:
This is implemented now :)

closed 2019-07-07 by AlessandroAU