AlessandroAU/Chorus32-ESP32LapTimer #32: Webserver - Error

mottex, 2019-03-11:
Hi Allesandro

I tried to get the webserver runnig, when im doing the sketch data upload - the following message will display:

Arduino: 1.8.7 (Windows 10), Board: "ESP32 Dev Module, Disabled, Minimal SPIFFS (Large APPS with OTA), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

[SPIFFS] data   : C:\Users\motte\OneDrive\Dokumente\Chorus Lap Timer\19-03-10 Chorus32-ESP32LapTimer-master\Chorus32-ESP32LapTimer-master\ESP32LapTimer\data
[SPIFFS] start  : 4001792
[SPIFFS] size   : 188
[SPIFFS] page   : 256
[SPIFFS] block  : 4096
/assets/css/images/overlay1.png

/assets/css/images/overlay2.png

/assets/css/images/overlay3.svg

/assets/css/images/overlay4.svg

/assets/css/main.css

/assets/css/slider.css

/err.html

/ESP32FS.java

/flag.svg

/GetStaticVars.js

/index.html

/index_.html

/jquery-3.3.1.min.js

SPIFFS_write error(-10001): File system is full.

error adding file!

SPIFFS Create Failed!
[SPIFFS] data   : C:\Users\motte\OneDrive\Dokumente\Chorus Lap Timer\19-03-10 Chorus32-ESP32LapTimer-master\Chorus32-ESP32LapTimer-master\ESP32LapTimer\data
[SPIFFS] start  : 4001792
[SPIFFS] size   : 188
[SPIFFS] page   : 256
[SPIFFS] block  : 4096
/assets/css/images/overlay1.png

/assets/css/images/overlay2.png

/assets/css/images/overlay3.svg

/assets/css/images/overlay4.svg

/assets/css/main.css

/assets/css/slider.css

/err.html

/ESP32FS.java

/flag.svg

/GetStaticVars.js

/index.html

/index_.html

/jquery-3.3.1.min.js

SPIFFS_write error(-10001): File system is full.

error adding file!

SPIFFS Create Failed!

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

AlessandroAU, 2019-03-13:
Hi mottex,

Firstly, you don't need the ESP32FS.java in the data directory. I assume you put it there by mistake when you were trying to install the ESP32 data uploading tool. You just need to put the .jar in the arduino program folder. Nothing else goes into the data folder. 

I think you are following old instructions, you are compiling with "Minimal SPIFFS (Large APPS with OTA)" , this was needed when we were using Bluetooth. With that option the file-space for the webserver is limited and the files don't fit, that is the error you see. Since BLE is disabled for now just compile with Partition Scheme => Default and reuploaded SPIFFS. It should work.

mottex, 2019-03-14:
Sorry was my fault! Thanks for your help

closed 2019-03-14 by mottex