AlessandroAU/Chorus32-ESP32LapTimer #33: Dev chat for the project

LuisDiazUgena, 2019-03-13:
Hi! 

I was thinking that it would be nice to have a chat (telegram or slack will be nice) in where the people who are developing the project could chat about it.  I think it will decrease dev time. 

I'm going to be coding a lot for the project and would love to have a chat room to discuss some stuff. 

@AlessandroAU Please, let me know what do you think about it.

ps915, 2019-03-13:
Telegram would be cool or Discord. Please no Slack! xD

LuisDiazUgena, 2019-03-13:
Telegram will be the best for me!

shaiss, 2019-03-14:
Discord please!

JyeSmith, 2019-03-14:
The problem with a chat room is the lack of audit trail in making decisions.  You'll see the Betaflight devs raise this issue and it's probably why the Pulls have a lot of comments.

While this project is substantially smaller, it's going to be easier to keep everything here in Issues.

2c

LuisDiazUgena, 2019-03-14:
I'm ok with whatever choice it's made.

It would be nice to have a roadmap of what is need to be do, and who is doing it.

JyeSmith, 2019-03-31:
@AlessandroAU we need Discord asap 😃

JyeSmith, 2019-04-04:
https://discord.gg/Ew3Gv3k