AlessandroAU/Chorus32-ESP32LapTimer #34: Connect to Chorus RF Laptimer

LuisDiazUgena, 2019-03-13:
Hi!

I've been trying to use the chorus RF laptimer App as listed on the README, and I can't see any of the info in the app.

I connect my android device to the Chorus32 network and signed into the network. I can see and change the setup. If I go to the Chorus RF Laptimer app, I can connect to the network (and it toast a "connected to 192.168.4.1". But there is nothing at all on the Freq and Pilots tabs. 

On Race there is two buttons "calibrate timers" and "start race", which are disabled.

I've seen that in comms.ino there is a lot of functions related to chorus API, so that make me think that it's already implemented, but I'm not sure if I'm missing something.

Thanks.

AlessandroAU, 2019-03-14:
Its working fine for me, what does your serial console output look like when you connect with the app?

AlessandroAU, 2019-03-14:
Luis, I just tried with a copy of your git and it works for me.

Perhaps your eeprom has bad values? Do you have numReceivers > 0?

LuisDiazUgena, 2019-03-14:
Wifi connection:

```
E (23967) event: mismatch or invalid event, id=63
E (23967) event: default event handler failed!
dhcps: send_offer>>udp_sendto result 0
```
After that, nothing happens when connecting to the app.

AlessandroAU, 2019-03-14:
That's really odd, can you try with another device? Maybe tell android to forget the network and try again?

LuisDiazUgena, 2019-03-14:
> That's really odd, can you try with another device? Maybe tell android to forget the network and try again?

Already did. Forgot network and removed app. Nothing changes (Android 9)

Instaled in other devide. Same behaviour (Android 7)

I'm sure that in both devices the app has all the permisions enabled.

AlessandroAU, 2019-03-14:
After you are directed to the captive portal webserver do you click 'ok' in android to accept the connection?

LuisDiazUgena, 2019-03-14:
> After you are directed to the captive portal webserver do you click 'ok' in android to accept the connection?

Got no message like that, but I can hit the menu icon and tell android to use the network.  The option pointed by the arrow translates as "Use this network at it is"

[image]

AlessandroAU, 2019-03-14:
If select 'use as is' and try the app again does anything change?

AlessandroAU, 2019-03-14:
Just checking,

In the app; I just want to check that you are clicking the 3 dots at the top right and then selecting 'connect via wifi' ?

LuisDiazUgena, 2019-03-14:
> Just checking,
> 
> In the app; I just want to check that you are clicking the 3 dots at the top right and then selecting 'connect via wifi' ?

Yes. (I work at Software developmnt so I know that you have to do this kind of questions, don't worry for asking)

LuisDiazUgena, 2019-03-14:
Just tried to connect again. Android just asked me:  This network has no internet connection, do you want to use it? Just clicked yes, and its works :)

closed 2019-03-14 by LuisDiazUgena