AlessandroAU/Chorus32-ESP32LapTimer #47: RSSI on RX4 spiking a bit with PCB v2

mike-levenick, 2019-04-03:
I've noticed that the ADC4 value on my new build is spiking around a bit. With nothing transmitting, RX 1-3 report RSSI bouncing between ~78-83ish, fluctuating by 5 or so. 

RX4 reports RSSI bouncing from 78 up to 110 or more at times, briefly, and then back down. I've also noticed the OLED will show that bar fill _completely_ (like, off the chart) for a brief moment and then settle down--although in-app RSSI monitoring does not show those massive spikes. 

After seeing similar interference from the jumper pads under the OLED, I'm wondering if it has something to do with a voltage rail near the ADC4 trace or something, since my point-to-point build does not exhibit this behavior. Especially since RX4 is powered from a different power rail than RX 1-3 are (the upper buck converter vs the lower one). 

I will be double checking all my solder joints, and also try a different ESP32 (I bought a 3 pack, and socketed the ESP32 into the board for easy swapping), but was wondering if anyone else has seen similar behavior. Perhaps a different type of filtering or something?

mike-levenick, 2019-04-03:
I will also try desoldering the lower power converter and jumping power over from RX3 to RX4 to see if performance is different.

AlessandroAU, 2019-04-04:
The massive spikes on the OLED are actually an artifact, in line 89 of OLED.ino we map the RSSI values to the sliders. To make best use of the space the minimum value of 50 is used. So if the RSSI drops below 50 the mapping function kind of breaks and it shows it as 100%.

Regarding the jumping around, I can't quite say yet, are you using the same RX5808 modules as your point-to-point setup?

JyeSmith, 2019-04-04:
@mike-levenick can you try swapping Rx 4 with one of the others and see if the problem follows?

mike-levenick, 2019-04-04:
@JyeSmith and @AlessandroAU I will troubleshoot on my end for the jumping RSSI. I believe it's likely hardware related on my end.

That said, I resolved the OLED artifacting for RSSI below 50 in PR48 https://github.com/AlessandroAU/Chorus32-ESP32LapTimer/pull/48

mike-levenick, 2019-04-05:
Resolved in commit https://github.com/AlessandroAU/Chorus32-ESP32LapTimer/commit/b3f9c253f969e188bc00d22fd3b7da0a042e4369

closed 2019-04-05 by mike-levenick