AlessandroAU/Chorus32-ESP32LapTimer #61: Multifunction use, long range omni diversity (passive antenna tracker)

Dropsleutel, 2019-04-25:
If the software could set all 6 modules, to the same Rx frequency and select the strongest output, via a video switch f.i. 2x CD4066, the unit could be also used as a long range multi diversity basestation with several hi gain antenna.

As here:
https://hobbyking.com/en_us/quanum-overlord-5-8ghz-40ch-6-antenna-diversity-receiver-w-raceband.html
Fi combined with these:
https://www.maartenbaert.be/quadcopters/antennas/triple-feed-patch-array-antenna/

mike-levenick, 2019-04-25:
I've looked into doing this as a separate project, but the hardware would be a fair bit different. You would need some sort of switching mosfet or relay to flip between the video signals based off RSSI. 

Definitely a project worth looking into, but I'm not sure if it's within the scope of Chorus32. Others may have other opinions tho.

Dropsleutel, 2019-04-25:
The analog video and audio switching there are simple solutions, it is far from complicated!
The easy way would be to use an 8 channel analog video switch (Mux) MAX4312, which is controlled via 3pins in BCD, see example here:

http://www.vwlowen.co.uk/radio/videoswitch3/index.htm
You could use 2 more of these switches for audio left and right.

AlessandroAU, 2019-04-25:
Its a good idea, and very doable. You can cut out 90% of the code. However, modern receivers have some 'other' fancy tricks like vsync reconstruction we can't currently do. 

I'm working on something to blow traditional RSSI switching RX's out of the water anyway.

http://www.eevblog.com/forum/projects/image-synthesizing-dsp-receiver-for-analog-fpv-video/

Dropsleutel, 2019-04-25:
hmm very interesting indeed :) Maybe this circuitry can be also of interest written in Dutch though:)
Called SyncSmurf, it is basically a PLL circuitry, mainly used for receiving weak and noisy ATV signals. 
It keeps, generates a strong sync even when signal is bad. 

Would be very nice to have a multipurpose unit, can be used for racetiming or for video reception etc

I hope project gets some more attention, because seriously nicejob!