AlessandroAU/Chorus32-ESP32LapTimer #101: USB plus Batt connect fries the ESP32

pascalnl, 2020-02-02:
Mistakenly I connected the battery when USB cabel was attached.
It fried the regulator on the ESP32

swampmonster, 2020-03-25:
That sucks man, never fun when hardware explodes. Not quite sure if you want help with something or perhaps want us to put a warning about this?

pascalnl, 2020-03-25:
Hi Alessandro,

No. More as a warning. Don't try this at home. Sort off thing.

I made a mistake.

Fortunately. The weather is getting better. So I can use it to get faster.

swampmonster, 2020-03-25:
Alright man. Good to hear you're gonna get some practice in soon :)
Also; I'm not Alessandro :)

LuisDiazUgena, 2020-04-06:
Hi,

I think it depends on what your setup is. I've connected like this, using both the battery and the usb and I got no problem at all.