AlessandroAU/Chorus32-ESP32LapTimer #104: Missing "brzo_i2c.h" file from compile

alvarobeiste, 2020-03-13:
Hello, I am getting the following error when trying to compile the project: 

C:\Users\AlphaFPV\Documents\Arduino\libraries\ESP8266_and_ESP32_OLED_driver_for_SSD1306_displays\src/SH1106Brzo.h:35:22: fatal error: brzo_i2c.h: No such file or directory

The libraries have been correctly installed, and the error happens both with the .zip lib or with the downloaded one.

Smeat, 2020-03-13:
I just used a fresh arduino 1.8.12 installation and followed the instructions in the readme without any problem.
Did you use version 4.1 of the library?

Can you post the complete error message? We are never using the file mentioned in the error, so no clue what went wrong.

alvarobeiste, 2020-03-13:
Hello!

I have just realized that I've used the Arduino IDE 1.8.9. I can confirm
that I did indeed use the 4.1 version of the library! I'll try again
tomorrow with the latest version, thank you.

Smeat, 2020-04-10:
@alvarobeiste Did you manage to fix the error you were getting?

alvarobeiste, 2020-04-10:
Yes, I forgot to close the issue! Reinstalled all the Arduino software and
libraries, tried again and got it done. Thanks for your help!

closed 2020-04-10 by Smeat