AlessandroAU/Chorus32-ESP32LapTimer #108: License of the project

Smeat, 2020-03-24:
We've never specified a license.
We are also required to attribute the original project (if we have any original code left).

Personally I'd prefer a license with copyleft (like the GPL), but would also be fine with the MIT license like the original project.

Whatever we choose, we need permission from all contributors.

Thoughts? @JyeSmith @AlessandroAU @LuisDiazUgena @mike-levenick @swampmonster

LuisDiazUgena, 2020-03-25:
Hi!

I've no problem to whatever license we use. Feel confortable with something like GPL as well.

swampmonster, 2020-03-25:
My personal preference would be for MIT/BSD but i have no objections to GPL if that's what the majority wants.

AlessandroAU, 2020-03-27:
I have no strong feelings either way, one thing though; if there is still MIT Licensed code in the repo is that still compatible with GPL?

swampmonster, 2020-03-27:
> I have no strong feelings either way, one thing though; if there is still MIT Licensed code in the repo is that still compatible with GPL?

A quick scan of https://www.gnu.org/licenses/license-list.html#GPLCompatibleLicenses would indicate that it's compatible. GNU in normal order wants to nitpick and says there is not such thing as an MIT license and points to the expat and x11 licenses as being the licenses called MIT but both of those are compatible as well so we should be in the clear on that point.

Smeat, 2020-03-27:
Yes MIT/X11 is no problem. This is one of the reasons I prefer the GPL: We can use code from almost any open source project we like :)

swampmonster, 2020-03-27:
If we go with the GPL we should probably at least talk to @voroshkov and see if he want's any of his code specifically marked as being MIT or if a general mention of where it came from will suffice.

Smeat, 2020-03-27:
If we still have code from the original project we must include the license header, so at least in Comms.cpp we are violating that.
Even with permission I'd like to keep (or rather add) the MIT text. Wouldn't feel right to omit that :)

JyeSmith, 2020-04-14:
I have no preference.  What ever keeps it open and free for anyone to use or commercialise.

mike-levenick, 2020-04-14:
I am good with whatever you all choose. :) no preferences.

PR #114 (2020-04-21): Set the license to GPLv3
Since only @swampmonster and me had a preference I decided to open a PR with the GPL ;P
Still open for arguments though.

Apart from personal preferences this would give us the most freedom in using code from other projects. There is no problem using MIT/X11 code and we can use GPL licensed code (like from Betaflight or pidflight-lap (another racetimer))

Closes #108 and solves the licencing issues with the original project.
closed 2020-04-21 by Smeat