AlessandroAU/Chorus32-ESP32LapTimer #113: Required hardware?

marcfon, 2020-04-10:
Can I just hook up an RX5808 to a ESP32 and run this project or do I need the PCB including all the other components?

By the way if there's a better place to ask questions like these please let me know!

marcfon, 2020-04-10:
Found the thread on [RCGroups](https://www.rcgroups.com/forums/showthread.php?2801815-Chorus-Laptimer-16-RSSI-5-8Ghz-track-system/)

closed 2020-04-10 by marcfon

Smeat, 2020-04-10:
The thread you linked is for the Chorus (no 32) project. The Arduino and ESP32 versions are two separate projects, which do support the same API, so all the client software is compatible between the two.
To answer your question: Yes an ESP32 and RX5808 is enough. You need some way to supply the required power though. The other components, like a display or buttons are completely optional.

We have a discord channel you can join: https://discord.gg/Ew3Gv3k