AlessandroAU/Chorus32-ESP32LapTimer #117: Wlan connection with activated Bluetooth not possible

Peschi90, 2020-04-15:
Hey Guys, 
thanks for your project but i have one issue.
because i cant connect with the app over wifi, i activated bluetooth.
This works ok with the App but if i want to change something in the WIFI settings Page,
the Module reboot directly after i connect to the wifi Network, closed the wifi,start it again and dont show the page.
do you know something about this issue?

BR Peschi :)

Smeat, 2020-04-15:
Are you using v1.0 or the current master? What else did you change in the configuration, besides activating bluetooth?
I had some problems with bluetooth the last time I tested it, so you might try and disable that to see if that fixes the problem

Peschi90, 2020-04-15:
iam using the current master but flashed it over arduino Software. When i deactivate bluetooth the i cann connet to the wifi and get the settings page. i get this issue only if i activate the bluetooth in the hardwareconfig.h file. 
the bluetooth connection is working with the app only the range is really poor.

Smeat, 2020-04-15:
Then I am guessing this is the same problem I encountered (https://github.com/Smeat/Chorus32-ESP32LapTimer/issues/7)
The Bluetooth stack is quite large and it seems we are running out of memory when we are using Bluetooth.
Unfortunately I can't say when to expect a fix for that. For now you'll have to disable Bluetooth.

Peschi90, 2020-04-15:
Memory could definetly be the problem. Sometimes if the Bluetooth connection lost because of range the chorus 32 stuck completely. I think you're right.