AlessandroAU/Chorus32-ESP32LapTimer #118: Compiling on Platformio?

marcfon, 2020-04-29:
The ESP32LapTimer.ino file says to check the readme but the information there is quite sparse :)

Any pointers in the right direction are appreciated!

mottex, 2020-04-29:
I need as well more information - how to use platformio!

s00500, 2020-04-29:
If you are running platformIO from the cli you can specify the env via the -e option, in case the port selection fails you can specify the port in the platformio.ini file for each environment
See here:
https://docs.platformio.org/en/latest/projectconf/section_env_upload.html#upload-port

`pio run -e BOARD_OLD -t upload`

s00500, 2020-04-29:
Or use --upload-port on to specify in the run command as well :-)
Hope I could help

marcfon, 2020-04-29:
I managed to build and upload the project using platformio using the steps described in the first post.

I haven't yet managed to specify the env target (BOARD_DEFAULT / BOARD_TTGO_LORA / BOARD_OLD) from the GUI. So need to use the CLI for now to build and upload.

marcfon, 2020-04-30:
This is how to compile Chorus32 with Bluetooth support...

1. Uncomment `#define BluetoothEnabled` in `ESP32Laptimer.cpp`
2. Uncomment `#define USE_BLUETOOTH` in `HardwareConfig.h`
3. Add this line to platformio.ini `board_build.partitions = huge_app.csv`

Smeat, 2020-04-30:
I won't recommend using Bluetooth. See #117

marcfon, 2020-05-01:
I'm sure one day bluetooth will be stable and this information might become useful :)

Also wifi is broken on app on Android 10 so bluetooth might be the only alternative for some.

Smeat, 2020-05-01:
But as of now, if you are using Bluetooth, the timer might crash any time, so I'd deem this unusable in the current state.

As for the Android problem: I should have a working fix by tommorrow

marcfon, 2020-05-01:
@Smeat Thanks for pointing that out. And looking forward to beta testing the android version :)

Pamilias, 2022-10-04:
do we still need to change to BOARD_OLD in HardwareConfig.h?

Smeat, 2022-10-04:
Depends on which hardware your are using:

> Use `BOARD_OLD` if you are using [this wiring schema](https://github.com/AlessandroAU/Chorus32-ESP32LapTimer/blob/master/img/schematic2.png) and `BOARD_DEFAULT` if you're using the [PCB v2.2 wiring schema](https://github.com/AlessandroAU/Chorus32-ESP32LapTimer/blob/master/pcb/JyeSmith/PCBV2.2/Schematic-V2.2.png).