AlessandroAU/Chorus32-ESP32LapTimer #119: Which build target matches the schematics?

marcfon, 2020-05-01:
I'm building a setup based on [this wiring schema](https://github.com/AlessandroAU/Chorus32-ESP32LapTimer/blob/master/img/schematic2.png).

Does this schema match the `BOARD_DEFAULT` or the `BOARD_OLD` build target?

Smeat, 2020-05-01:
BOARD_OLD.
BOARD_DEFAULT is the PCB from Jye: https://github.com/AlessandroAU/Chorus32-ESP32LapTimer/blob/master/pcb/JyeSmith/PCBV2.2/Schematic-V2.2.png

marcfon, 2020-05-02:
Thanks! Is there a change log kind of thing to see how what changed in BOARD_DEFAULT? For instance if you want rewire things to match the new board.

marcfon, 2020-05-02:
What is the purpose of the two separate oled connections from vcc and gnd on the v2.2 PCB?

Smeat, 2020-05-02:
> Thanks! Is there a change log kind of thing to see how what changed in BOARD_DEFAULT? For instance if you want rewire things to match the new board.

No just a different design for a PCB. Since most digital functions can be done by any pin, the assignment is quite arbitrary.
If you already have wired everything, there is not much reason to change that.

> What is the purpose of the two separate oled connections from vcc and gnd on the v2.2 PCB?

I think it is to accommodate for different pinouts with those OLED modules

marcfon, 2020-05-03:
After some trail and error I found the [v2.2 wiring easier](https://raw.githubusercontent.com/AlessandroAU/Chorus32-ESP32LapTimer/master/pcb/JyeSmith/PCBV2.2/Schematic-V2.2.png) to match with the **ESP32 Devkit v1** that I'm using. With the first wiring I used you need to translate the pinout (and I failed to do that successfully).