AlessandroAU/Chorus32-ESP32LapTimer #123: PlatformIO (VSCode)

TechVic-1, 2020-07-07:
Move Platformio.ini and .gitignore out of ESP32LapTimer folder

Smeat, 2020-07-07:
If we move `platformio.ini` we break the build system ;)
What is the reason you want to move it?

TechVic-1, 2020-07-07:
NO you don't. This is the first repository that doesn't have the platformio.ini file in root folder. Check any repository that uses vscode and platformIO you will see. I want to move it because platformio doesn't like when it's not in root and using  GitHub Desktop then you can't use the built in functions like build and upload. Because platformio doesn't recognize it as an project

TechVic-1, 2020-07-07:
Example: [MarlinFirmware](https://github.com/MarlinFirmware/Marlin)

Smeat, 2020-07-07:
Without modification it'll break, because it won't find the `src` folder.
As this project also contains other stuff besides the firmware, I don't see any reason or benefit to move it to the root. Just seems like more clutter.
It also prevents us from adding other firmwares, which might be the case in the near future. Not clue if there is a clean way to do it, since my pio knowledge is rather limited.

Besides that I am still missing the point. If your program won't recognize the project, can't you just select "ESP32LapTimer" as the root? (Sorry I have absolutely no clue about the software you are using)
I think there were some people using it, but the documentation is still TODO.

If you can make a compelling argument why this is beneficial, feel free to submit a pr.
But "others do it" isn't a valid argument. A quick search even showed, that there a bunch of projects, which don't have the platformio.ini in their root.

TechVic-1, 2020-07-08:
Yes Okay you need to add the lines in platformio.ini `[platformio]
src_dir      = ESP32LapTimer`
Check out FireFly-Nano platformio.ini file, here you just uncomment which board you have and hitt build or upload.
I just wanted to help make it easier for people to compile it

TechVic-1, 2020-07-08:
[GitHub desktop](https://desktop.github.com/)

Smeat, 2020-07-08:
> I just wanted to help make it easier for people to compile it

Nothing wrong with that :) I am always happy to get a different perspective, since my knowledge with the GUI-tools is rather limited.
I think in this case it is just a problem of the missing documentation.

Having to edit the platformio.ini is something I didn't do by design, since editing the config files of the build system, just to select a different target, is definitely the wrong approach and runtime configuration should always be preferred.

s00500, 2020-08-24:
to select the board environments should be used