AlessandroAU/Chorus32-ESP32LapTimer #124: INA219 incorrect reading

marcfon, 2020-07-16:
The reading from my INA219 off by about 0.3v compared to when I read voltage with a multimeter across the battery + and -.

I've got the VIN- connected to + terminal of the battery [according to this schema](https://github.com/AlessandroAU/Chorus32-ESP32LapTimer/blob/master/img/schematic2.png).

Did a miss a wire or setting somewhere?

Smeat, 2020-07-21:
Sorry, I don't have an INA219, so my ability to help is limited.
But it seems the wiring diagram is missing something, as VIN- is unconnected. You better use [Jye's PCB](https://raw.githubusercontent.com/AlessandroAU/Chorus32-ESP32LapTimer/master/pcb/JyeSmith/PCBV2.2/Schematic-V2.2.png)  as a reference.
Also make sure you are measuring as close to the INA219 as possible.

marcfon, 2020-07-21:
Thanks for thinking along Smeat. You don't have to have VIN- connect if you're just monitoring voltage. If you also want to measure current then you do so it depends. I fiddled around a bit more but couldn't find the cause. Not a dealbreaker but curious if more people ran into this.