AlessandroAU/Chorus32-ESP32LapTimer #127: Compilation error 'touch_pad_filter_start' was not declared in this scope

arnaudmorin, 2022-06-04:
Hello,

I cloned the code
Entered the `ESP32LapTimer` folder
Then trying to upload:

```bash
pio run -e BOARD_OLD -t upload

src/Buttons.cpp: In function 'void newButtonSetup()':
src/Buttons.cpp:63:3: error: 'touch_pad_filter_start' was not declared in this scope
   touch_pad_filter_start(BUTTON1);
   ^~~~~~~~~~~~~~~~~~~~~~
src/Buttons.cpp:63:3: note: suggested alternative: 'touch_trigger_src_t'
Compiling .pio/build/BOARD_OLD/src/HardwareConfig.cpp.o
   touch_pad_filter_start(BUTTON1);
   ^~~~~~~~~~~~~~~~~~~~~~
   touch_trigger_src_t
src/Buttons.cpp:64:3: error: 'touch_pad_set_filter_period' was not declared in this scope
   touch_pad_set_filter_period(BUTTON1);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~
Compiling .pio/build/BOARD_OLD/src/Laptime.cpp.o
*** [.pio/build/BOARD_OLD/src/Buttons.cpp.o] Error 1

```

arnaudmorin, 2022-06-04:
After removing the faulty buttons.cpp code:

```bash
$ git diff

diff --git a/ESP32LapTimer/src/Buttons.cpp b/ESP32LapTimer/src/Buttons.cpp
index 22621a7..db3205d 100644
--- a/ESP32LapTimer/src/Buttons.cpp
+++ b/ESP32LapTimer/src/Buttons.cpp
@@ -59,11 +59,6 @@ void newButtonSetup() {
 #ifdef USE_NORMAL_BUTTONS
   pinMode(BUTTON1, INPUT_PULLUP);
   pinMode(BUTTON2, INPUT_PULLUP);
-#else
-  touch_pad_filter_start(BUTTON1);
-  touch_pad_set_filter_period(BUTTON1);
-  touch_pad_filter_start(BUTTON2);
-  touch_pad_set_filter_period(BUTTON2);
 #endif
 }

```

it compiles, but finally error because of too big firmware:

```bash
Error: The program size (1448665 bytes) is greater than maximum allowed (1310720 bytes)
```
What should I do?

Z1gun, 2022-08-01:
I have pio 6.1.3 and got same issues on Windows and Linux machine. What the correct solution of Buttons.cpp compiler error?

Smeat, 2022-08-01:
I was using pip. Sorry, forgot to add that. Fixed it in my previous post.

> What the correct solution of Buttons.cpp compiler error?
That is the weird part: I don't get any error on my machine.

But I was able to reproduce both in a VM. It is probably some library update. I'll look into it in the following days.

Smeat, 2022-08-20:
This should be fixed with #130. If not feel free to reopen.

PR #130 (2022-08-20): Force old version of espressif32
The new version changes a bunch of things. Version 3.5.0 should work fine for now.

This should fix #127. @Z1gun can you confirm this fixes your issue?
closed 2022-08-20 by Smeat